intel/systemc-compiler #1: Building and installing ICSC errors

minamagdy00, 2020-12-14:
Hi,

I'm running a manual installation for ICSC and could build and install Protobuf and LLVM/Clang successfully.
But when building ICSC i got the following errors:

[ 72%] Building CXX object sc_tool/CMakeFiles/SCTool.dir/lib/sc_tool/expr/ScParseExpr.cpp.o
In file included from /s/mimagdy/test/icsc/sc_tool/lib/sc_tool/cfg/SValue.h:15:0,
                 from /s/mimagdy/test/icsc/sc_tool/lib/sc_tool/cfg/SValue.cpp:12:
/s/mimagdy/test/icsc/sc_tool/lib/sc_tool/diag/ScToolDiagnostic.h:47:24: error: expected ‘;’ at end of member declaration
     const char* what() const 
                        ^
/s/mimagdy/test/icsc/sc_tool/lib/sc_tool/diag/ScToolDiagnostic.h:48:5: error: ‘_GLIBCXX_TXN_SAFE_DYN’ does not name a type
     _GLIBCXX_TXN_SAFE_DYN _GLIBCXX_USE_NOEXCEPT override {
     ^
/s/mimagdy/test/icsc/sc_tool/lib/sc_tool/diag/ScToolDiagnostic.h:47:17: error: looser throw specifier for ‘virtual const char* sc::InternalErrorException::what() const’

I got to know that it may be a c compiler issue but i'm using the recommended versions:
-- The C compiler identification is GNU 4.8.5
-- The CXX compiler identification is GNU 4.8.5
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found LLVM 7.0.0

-- CMAKE_CXX_STANDARD = 17
-- CMAKE_CXX_STANDARD_REQUIRED = 
-- CMAKE_SYSTEM = Linux-3.10.0-693.21.1.el7.x86_64
-- CMAKE_SYSTEM_PROCESSOR = x86_64
-- QT_ARCH = x86_64
-- Threads_FOUND = TRUE
-- CMAKE_USE_PTHREADS_INIT = 1
-- CMAKE_THREAD_LIBS_INIT = -pthread

ripopov, 2020-12-22:
Gcc 4.8 won't work. Try more modern compiler: I was able to build with gcc 9.1 and clang 7

minamagdy00, 2020-12-22:
> Gcc 4.8 won't work. Try more modern compiler: I was able to build with gcc 9.1 and clang 7

Thanks ripopv.
I used/installed GNU 10.2 and it worked.

-- The C compiler identification is GNU 10.2.0
-- The CXX compiler identification is GNU 10.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done

closed 2021-02-18 by mikhailmoiseev