intel/systemc-compiler #4: Installation and Uninstallation

Z-KN, 2021-03-07:
A very nice work! But I still have two questions.
(1) If I have installed protobuf and llvm before, but not in path $ICSC_HOME, can I skip steps of installating them that written in the install.sh and manual installation guide?
(2) How can I uninstall ICSC?

mikhailmoiseev, 2021-03-07:
1) If you have LLVM 7.0.0 with Clang already installed it could be potentially used. ICSC uses LLVM and Protobuf packages which should be found by cmake during installation. Next ICSC versions will be based on newer versions of LLVM/Clang, so your local installation needs to be upgraded. I would recommend to go with install script provided.
2) Just delete $ICSC_HOME folder.

mikhailmoiseev, 2021-03-12:
User guide is not ready yet. Planning to commit it in the end of March. 
Most important details are given at [SystemC/C++ supported](https://github.com/intel/systemc-compiler/wiki/SystemC--supported).

mikhailmoiseev, 2021-04-03:
User guide document added in /doc folder.

closed 2021-04-03 by mikhailmoiseev