intel/systemc-compiler #5: Please avoid nonblocking assignments in always_comb blocks

ShashankVM, 2021-03-23:
With reference to the SystemC Fika 2021, I noticed that there were some nonblocking assignments found in generated always_comb blocks. While this may not cause a problem for synthesis, it will definitely create problems in simulation.

mikhailmoiseev, 2021-03-23:
What kind of problem in simulation do you have? Could you please give an example. 
What simulation tool you are using?

mikhailmoiseev, 2021-03-23:
I have checked the tool logic and some tests and could not find non-blocking assignment in always_comb. Could you point to the example where you found that?

ShashankVM, 2021-03-23:
It was on slide 7 of the presentation on SystemC Fika

mikhailmoiseev, 2021-03-23:
It is my mistake, manually created example, sorry for that. The tool does not generate any non-blocking assignment in always_comb. Thank you for help.

ShashankVM, 2021-03-23:
If the tool does not create nonblocking assignment in always_comb then I'll close this issue. Thank you.

closed 2021-03-23 by ShashankVM