intel/systemc-compiler #7: The use of new operators

rseac, 2021-05-25:
Some of the tests seem to use `new`, `sc_new<>`, `sc_new_array<>`.  It seems that the synthesizable reference suggests that `new` can be used only for SC_MODULE types and arrays.  However, some tests show it being used for dynamically allocating ports (`new sc_out<bool>("p")`). Does systemc-compiler go beyond what is required by the reference?  

What is the purpose of `sc_new` and `sc_new_array`?

mikhailmoiseev, 2021-05-26:
Allocation using ```new``` is supported for ```sc_object``` inheritors (ports, signals, modules). For all other cases ```new``` is not supported, use ```sc_new``` for single object and ```sc_new_array``` for array. See several examples [here](https://github.com/intel/systemc-compiler/wiki/SystemC--supported#sc_new-and-sc_new_array-declarations). All details of supported SystemC are given [here](https://github.com/intel/systemc-compiler/wiki/SystemC--supported).

ICSC uses dynamic elaborator which cannot distinguish between pointer to dynamically allocated object and dangling pointer. We have solved this problem by overriding operators ```new``` and ```new[]``` in the pathed SystemC library. This allows to trace all memory allocations that take place during elaboration. For other SC and C++ types it needs to use ```sc_new``` and ```sc_new_array```. It is planned to fix that with using information of pointers from code analysis, hopefully it will be done this year.

rseac, 2021-05-26:
It's great to have all that support. I noticed in the synthesizable reference it says "The new operator (ISOC++ 5.3.4) has Restricted Support. It is Supported for the instantiation of variable arrays and SC_MODULE. It is Not Supported for the instantiation of types derived from the sc_object type, other than SC_MODULE. "  This seems to suggest that what systemc-compiler provides is greater than what is specified in the reference.

mikhailmoiseev, 2021-05-28:
Yes, ICSC supports arrays of modules/signals/ports and arrays of pointers to them.  Arrays can be module members and function local variables. 
I prefer to use ```sc_vector``` instead of array of modules/signals/ports, as it provides unique names for instances and allows to bind all its elements w/o loop. Currently, there is a limitation for ```sc_vector``` of modular interfaces (module inherits ```sc_interface```) which will be fixed soon.

mikhailmoiseev, 2021-12-31:
```sc_vector``` of modular interfaces supported now, so close this issue.

closed 2021-12-31 by mikhailmoiseev