intel/systemc-compiler #8: const member parameter error

jinjie320621, 2021-05-28:
i define one const member parameter (**const unsigned int tRCD**; ) in class Test (example), class Test is a modular interface. Then use class Test as: Test t_test[10]. 
After SCT operation, i see the sv_out/.sv file, the code as:
**localparam logic [31] tRCD = `d22;** //on the top module, because Test is the modular interface

In each t_test[i]，I think it should use tRCD the whole 32bit value. But it use as:
//example 
**t_wait_time = tRCD[i];// i think it shoud be  t_wait_time = tRCD here.**

Please help me check whether it is a limitation for ScTool? Thanks very much.

mikhailmoiseev, 2021-05-28:
Thank you for feedback, that looks as a bug. Will check and update near days.

mikhailmoiseev, 2021-06-14:
This issue has been fixed in 1.3.26. Please check and close if it works for you.

jinjie320621, 2021-06-21:
Issue has beed fixed, thanks very much.

jinjie320621, 2021-06-21:
Issue has been fixed, thanks.

closed 2021-06-21 by mikhailmoiseev