intel/systemc-compiler #9: Example SystemC simulation is hanging and the instructions for simulation does not work

ShashankVM, 2021-06-17:
Greetings,

1. There is a minor error in the instruction for SystemC simulation. I will make a pull-request to fix that.
2. The simulation is hanging. I have fixed that and enhanced the testbench. This will be included in my pull-request.

ShashankVM, 2021-06-18:
Currently I have closed my pull-request since users might remove the build directory. I will check again and reopen my pull-request.

ShashankVM, 2021-06-18:
@mikhailmoiseev  are pull requests welcome?

ShashankVM, 2021-06-18:
Hi @mikhailmoiseev : The Linux commands in the documentation don't work. So I made a pull request to fix it. I've also modified the counter testbench so that it stops and prints the output values.

mikhailmoiseev, 2021-06-18:
Hi @ShashankVM. Thank you for your effort. I will check your pull request near weeks.

closed 2021-06-21 by mikhailmoiseev