intel/systemc-compiler #13: const parameter can not support sc_biguint

jinjie320621, 2021-09-30:
When I use sc_biguint as type of one const member parameter, which data bits more than 64, will cause error.

One example, const sc_biguint<65> data;
error: Unsupported type: 'const sc_dt::sc_biguint<65>'

Is this a limitation of this tool?
Thanks very much.

mikhailmoiseev, 2021-10-02:
Yes, it is limitation described at [SystemC supported](https://github.com/intel/systemc-compiler/wiki/SystemC--supported): 
_45. Constant fields can be initialized in module constructor initialization list and body, they are generated as localparam in SV. Constant of any integral type initialized at elaboration phase should be 64 bit or less (stored in uint64/int64 field). Constant more than 64 bit has unknown value after elaboration, therefore error is reported._ 

Potentially, it is possible to support >64 constants initialized with concatenation operator:
```cpp
const sc_uint<64> A = 0x8000000000000000;
const sc_uint<64> B = 0x8000000000000000;
const sc_biguint<128> C = (A, B);  -- error reported now
``` 
```verilog
localparam logic[63:0] A = 64'h8000000000000000;
localparam logic[63:0] B = 64'h8000000000000000;
localparam logic[127:0] C = {A,B};
```
Will add this to future plans. Is that critical for your designs or it is just checking tool evaluation?

mikhailmoiseev, 2022-04-12:
Since current version SC type variable can be initialized with string literal or string variable which **support >64 bit values**. 

```cpp
const char* cstr = "43";
std::string str;                 // Initialized in constructor or a method call

void someProc() {
    sc_uint<16> ux = "0b110011"; // Initialization with string literal
    ux = "42";                   // String literal assignment    
    ux = sc_uint<5>("0x42");     // Cast at initialization supported
    ux = cstr;                   // C string assignment
    ux = str.c_str();            // std::string assignment
    sc_biguint<65> bu;  
    bu = "0x1FFFF1111FFFF1111";  // More than 64bit literal
}

jinjie320621, 2022-05-14:
Got it, thanks very much.
I must say sorry for close this issue so later.

closed 2022-05-14 by jinjie320621