intel/systemc-compiler #18: Connecting SystemC compiler to open source ASIC/FPGA tooling and demo with Google's Skywater PDK?

mithro, 2022-03-04:
Thanks to the funded provided by DARPA there is an increasingly capable, fully automated, RTL to GDS toolchain for creation ASICs called OpenROAD (https://theopenroadproject.org/). To help support the project, Google has partnered with SkyWater to release a fully open source, manufacturable 130nm PDK (https://github.com/google/skywater-pdk) and with efabless a run a regular no-cost MPW program for open source designs.

Google has also been actively working with Antmicro on improve the state of SystemVerilog tooling in the open source tooling with projects now being hosted by CHIPS Alliance like;
 * https://opensource.googleblog.com/2021/09/open-source-systemverilog-tools-in-asic-design.html
 * https://github.com/chipsalliance/sv-tests
 * https://github.com/chipsalliance/Surelog
 * https://github.com/chipsalliance/UHDM -- Library and connection to tools like Yosys and Verilator
    * https://antmicro.com/blog/2020/12/ibex-support-in-verilator-yosys-via-uhdm-surelog/
    * https://antmicro.com/blog/2021/12/coroutines-for-dynamic-scheduling-in-verilator/
    * https://antmicro.com/blog/2022/02/simplifying-open-source-sv-synthesis-with-the-yosys-uhdm-plugin/
 * https://github.com/chipsalliance/verible -- Developer tooling for SystemVerilog like an auto-formatter, linter and library to create more!
   * https://antmicro.com/blog/2021/08/verible-integration-with-github-actions/

There is also a bunch of work around fully open source RTL to Bitstream flows here too -- see https://chipsalliance.org/announcement/2022/02/18/chips-alliance-forms-f4pga-workgroup-to-accelerate-adoption-of-open-source-fpga-tooling/

It would be **_awesome_** to have your SystemC compiler as a frontend for these tools so that you could have a complete open source SystemC to GDS or SystemC to bitstream flows. It would be even more awesome if someone used that to then tape out a real ASIC on the no-cost MPW program.

Due to everything here being open source you could even have an end-to-end demo for SystemC to GDS which results in an actually manufacturable IC without need any set up!

mithro, 2022-03-04:
PS Intel is a member of the CHIPS Alliance where a lot of this work is happen!

mikhailmoiseev, 2022-03-06:
Yes, it would be interesting to integrate the SystemC compiler into an open source design flow. That lets universities and individual researchers to build own silicon in a simple way.
Do you suggest any steps to collaborate?

mithro, 2022-03-06:
Send me an email at tansell@google.com and we can discuss options?

I think the first step would be to see if output from your compiler can be parsed by the current open source SystemVerilog frontends. I have a diagram at https://j.mp/sv-flow-diagram which kind of shows the options. It might also be worth looking at getting the output into the SystemVerilog test suite @ https://github.com/chipsalliance/sv-tests

closed 2022-10-27 by mikhailmoiseev