intel/systemc-compiler #20: Primitive channel

Spoorthy-Siddannaiah, 2022-04-11:
I tried to synthesize a module that has a primitive channel. I used an example simple_fifo which is the repository, https://github.com/intel/systemc-compiler/tree/main/systemc/examples/sysc/simple_fifo. 
But I am getting an error stating " Static record is not supported yet "

fatal error: ScTool internal fatal error : Static record is not supported yet
1273:     static const sc_event none;
1273:     ^
1273: ----------------------------------------------------------------
1273:  SystemC-to-Verilog trasnlation, ERROR 
1273: ----------------------------------------------------------------
1273: 1 error generated.
1273: Error while processing /home/spoorthy/work/systemc-compiler/build/icsc/designs/Simple/Simple.sctool.cpp.
1273: Top-level module is top
2/2 Test #1273: Simple_SYN .......................***Failed    2.14 sec

The following tests passed:
        Simple_BUILD

50% tests passed, 1 tests failed out of 2

I want to know how to synthesize the module with the primitive channel. Can you please guide me with this?

mikhailmoiseev, 2022-04-12:
The example you use contains sc_event (line 91) which is not synthesizable according with SystemC synthesizable standard.
Please find a number of synthesizable examples in https://github.com/intel/systemc-compiler/tree/main/examples.
It is prohibited to synthesize primitive channel and its inheritors. If you want to create a kind of FIFO or another channel, it could be based on sc_signal channel and has sc_in/sc_out ports. 

Some examples and explanations how to prepare SystemC design to synthesis are given in https://github.com/intel/systemc-compiler/blob/main/doc/ug.pdf

mikhailmoiseev, 2022-04-12:
There is more or less general implementation of FIFO in https://github.com/intel/systemc-compiler/blob/main/components/common/sctcommon/sct_fifo.h which is synthesizable.

closed 2022-05-03 by Spoorthy-Siddannaiah