intel/systemc-compiler #21: Question regarding missing SystemC features

maubri, 2022-04-19:
Hi,

First of all, thanks for starting this great project I think this is really useful.

I just really have questions regarding missing features, or rather perhaps the ones I mostly care about and what the plans are for those. These are:

1. Sequential Methods support
2. Reading outputs within a sequential process
3. Records:
  a. Support for base class
  b. Support for record in record
  c. Support for record as signal/port template argument
4. Support for [ac_int](https://github.com/hlslibs/ac_types/blob/master/include/ac_int.h). These types have many advantages over sc_dt ones and seem to be reasonably popular in HLS scope

I'm not sure 2 above is actually illegal, I think it might be ok but then I believe it is usually not deemed a good practice. If one would like to address that, for every module output one could instantiate a module-local logical, assign the output to this local logical (with an assign statement) and then use the latter inside processes to read/write freely instead of the output.

Anyway, my question is whether there's some kind of roadmap line up for this project that may address some of these? Would you accept contributions regarding any of these, and what would be the best way to proceed.

Many thanks!

mikhailmoiseev, 2022-04-20:
Hi,

1. Sequential SC_METHOD can be represented with SC_CTHREAD. Do you see any advantages of such SC_METHOD?
2. Outputs (sc_out<>) can be read in SC_CTHREAD.
3. Record support is limited due to high complexity of the implementation. Record for signal/ports may be planned for this year. Record in record and record as base class not planned to support.
4. ICSC is focused on SC synth subset standard, no external types are planned to be supported. Which tools except Catapult support ac_int? 

There is an internal roadmap and the main internal repository to contribute inside of Intel. I am considering to move all that to here near time. Currently, as this repo is just a mirror of the internal one, only bug fixes and external SC designs can be accepted.

maubri, 2022-04-21:
Hi,

Thanks for your answers.

1. Seq SC_METHOD are a full part of the SystemC synth spec (chapter 4.1.2 in the spec). An advantage of going for them instead of SC_CTHREAD is SC_METHOD run slightly faster within a SystemC simulation. So there's an advantage in terms of simulation time specially for large designs
2. Yes indeed. My suggestion was maybe just _cosmetic_
3. Nice to hear record as signal/port argument will get supported. Regarding record in record and base class I understand it is more complex, maybe something could eventually be done if C++ classes' storage are straight/blindly translated to SV packed structs. Maybe even a hint to the AST parsing could be provided by the user, for example if all classes to be translated to SV should derive from a translate_to_sv base class
4. I'm not sure about other tools aside from Catapult. I think at some point ac_types were proposed to be included as part of SystemC, not sure what happened with that. Otherwise this [Intel product](https://www.intel.com/content/www/us/en/develop/documentation/oneapi-fpga-optimization-guide/top/quick-reference/algorithmic-c-data-types.html) seems to support them? I personally use them because they are much faster than sc_bigint in particular with wide types (>64). Another big advantage of ac_int over sc_dt is the issues of mixing unsigned and signed have been addressed, you can check out this in the documentation of ac_int

> There is an internal roadmap and the main internal repository to contribute inside of Intel. I am considering to move all that to here near time. Currently, as this repo is just a mirror of the internal one, only bug fixes and external SC designs can be accepted.

Many thanks, I look forward to this and knowing more about the roadmap.

mikhailmoiseev, 2022-04-21:
Thank you, for explanations. 

Will add sequential SC_METHOD support in near plans.
As for AC types vs SC types, yes there is a problem with bigint/biguint performance, which is planned to be solved by SC LWG. There is a fast implementation which, I hope, will be available soon. 
Signed/unsigned mixing of SC types is real headache, so will consider to support AC types in some future. 

If you have any examples/designs which are passed through the ICSC, please share the experience or the code if possible.

mikhailmoiseev, 2022-04-28:
Hi,

In the Version 1.4.21 records with base class(es) and records with member records are supported. Multiple inheritance supported. There is main limitation: inner/base record constructor body must be empty. See more details at https://github.com/intel/systemc-compiler/wiki/SystemC--supported

Please let me know if you find any bugs/issues there.