intel/systemc-compiler #23: multiple level mif array cause error

jinjie320621, 2022-05-14:
I design 3 modules: TestA, TestB, and TestC.
Test B and TestC are mif.

3 level hierarchy architecture: TestA is the top module, TestB is TestA’s child module, TestC is TestB’s child module.
TestB is instantiated as a mif array in TestA, TestC is instantiated as a mif array in TestB.

It will cause ICSC tool error: ScTool internal fatal error : **_No record in getBaseClass()_**

But when not instantiate TestC (or TestC is not a mif array), it means not have multiple level mif array, it will be correct.

Is it a bug of ICSC tool? How can I avoid it? Thanks very much.

mikhailmoiseev, 2022-05-20:
Hello,

Thank you for this issue.

Currently inner MIF (MIF inside MIF) is not supported. You can use normal child module (not MIF) without limitations.

Would also recommend to use ```sc_vector``` for SystemC modules, signals, ports instead of C++ array.

Will update this issue when/if inner MIF supported.

mikhailmoiseev, 2022-05-26:
That has been supported for ```sc_vector``` in Version 1.4.25. 
See tests 
https://github.com/intel/systemc-compiler/blob/main/tests/mif/test_sc_vector_mif5.cpp and 
https://github.com/intel/systemc-compiler/blob/main/tests/mif/test_sc_vector_mif6.cpp

Please consider C++ array of MIF will not be supported.

mikhailmoiseev, 2022-08-01:
As soon there is no comments, consider as done.

closed 2022-08-01 by mikhailmoiseev