intel/systemc-compiler #36: Can tlm interfaces be used?

orichal, 2022-07-20:
For example:

Hierarchy:
----------
TOP
 --> TB
 --> DUT

DUT module:
-----------
sc_port< tlm_nonblocking_get_peek_if< bool > > status;
...
if (status->nb_can_get())
{
    bool status_in;
    status->nb_get(status_in);
}

TB module:
----------
sc_port< tlm_nonblocking_put_if< bool > > status;
...
if (status->nb_can_put())
{
    bool status_out = true;
    status->nb_put(status_out);
}

TOP module:
-----------
DUT dut;
TB  tb;
tlm_fifo <bool> chan_status;
...
tb.bind(chan_status);
dut.bind(chan_status);

mikhailmoiseev, 2022-07-20:
TLM 1.0 interfaces are not supported. 

There is a communication library with blocking/non-blocking interfaces similar to TLM 1.0 which are synthesizable. The library also has external ports with TLM 2.0 interfaces. It will be available near months.

orichal, 2022-07-21:
> TLM 1.0 interfaces are not supported.
> 
> There is a communication library with blocking/non-blocking interfaces similar to TLM 1.0 which are synthesizable. The library also has external ports with TLM 2.0 interfaces. It will be available near months.

Thank you for a quick answer.
Is the TLM-1.0 equivalent part of the communication library available ?

mikhailmoiseev, 2022-07-21:
The communication library modules have sct_put_if/sct_get_if interfaces given below.  The reason why TLM 1.0 was not taken is it includes too many functions and some of them cannot be synthesized. 

| Interface  | Functions | Comment |
| ---------- | --------- | ------- |
| sct_put_if<T>  | ```bool ready()``` | Return true if it is ready to put request |
|                | ```void reset_put()``` | Reset this initiator/FIFO |
|                | ```void clear_put()``` | Clear (remove) request put in this cycle |
|                | ```bool put(const T& data)``` | Put request into initiator/FIFO if it is ready, return ready to request |
|                | ```bool put(const T& data, sc_uint<N> mask)``` | Put request into initiator/FIFO if it is ready, ```mask``` used to enable/disable put or choose targets in multi-cast put, return ready to request |
|                | ```void b_put(const T& data)``` | May-block put request, could be used in THREAD process only  |
|                | ```bool has_put()``` | Put request done at last DC, combinational in RTL mode |
|                | ```void addTo(sc_sensitive& s)``` | Add put related signal to process sensitivity | 
|                | ```void addHasPutTo(sc_sensitive& s)``` | Add signals to sensitivity for process which calls ```has_put()``` | 
| sct_get_if<T>  | ```bool request()``` | Return true if it has request to get |
|                | ```void reset_get()``` | Reset this target/FIFO |
|                | ```void clear_get()``` | Clear (return back) request got in this cycle |
|                | ```T peek()``` | Peek request, return current request data, if no request last data returned |
|                | ```T get()``` | Get request and remove it from FIFO/target, return current request data, if no request last data returned |
|                | ```bool get(T& data, bool enable)``` | Get request and remove it from FIFO/target if ```enable``` is true, return true if there is a request | 
|                | ```T b_get()```  | May-block get request, could be used in THREAD process only |
|                | ```bool has_get()``` | Get request done at last DC, combinational in RTL mode |
|                | ```void addTo(sc_sensitive& s)``` | Add get related signal to process sensitivity | 
|                | ```void addPeekTo(sc_sensitive& s)``` | Add peek related signal to process sensitivity | 
|                | ```void addHasGetTo(sc_sensitive& s)``` | Add signals to sensitivity for process which calls ```has_get()``` |

orichal, 2022-07-24:
Thank you, seems as it covers the most important part of TLM 1.0 nb/b get/put/peek . Looking forward!

mikhailmoiseev, 2023-05-20:
Please find the SingleSource library with TLM interfaces implementation at https://github.com/intel/systemc-compiler/tree/main/components/common.
There is description of the library channels in update UserGuide: https://github.com/intel/systemc-compiler/blob/main/doc/ug.pdf

closed 2023-09-30 by mikhailmoiseev

orichal, 2023-10-04:
Hi Mikhail,

Thank you very much.

Ori