intel/systemc-compiler #42: Examples dvcon20 fifo not working as expected

nico1507, 2022-08-31:
Hi there,
I looked into the **dvcon20** fifo example. Seems like there is an issue with the valid and/or ready signals of the fifo(s) within **AdvFifo.h**. When modifying the **test_fifo.cpp** by intruducing additional wait cycles into either the `reqProc()` or `respProc()` of the Dut, the expected behaviour is waiting of the other process. Both SC_CTHREADS check the fifo for valid/ready before they read/write data from/to it. 

```
  void reqProc() {
      wait();
      
      while(true) {
          if (fifo.ready()) {
              fifo.push(in.read());
          }
          wait();
      }
  }
  
  void respProc() {
      wait();
      
      while(true) {
          if (fifo.valid()) {
              out = fifo.pop();
          }
          wait();
      }
  }
```

What actually happens is multiple readouts of the same value or missing values respectively. The test within the testbench is also not sufficient to ensure the functionality of the fifo. A counter is used as an input and it is only checked if the output of the fifo is not larger than its input.
 
```
  for (int i = 0; i < N; i++) {
      in = i;
      wait();
      
      if (out.read() > in.read()) {
          cout << "in " << in.read() << " out " << out.read() << endl;
          assert (false && "error");
      }
  }
```
A fifo should output the exact same values as it received before. The testbench should be adapted to ensure this behaviour and detect multiple readouts of the same value or missing values.

mikhailmoiseev, 2022-09-02:
Thank you for the issue found. Will check that.

mikhailmoiseev, 2022-09-03:
The current implementation of ```adv_fifo``` has synchronously de-asserted valid/ready/afull outputs as described in the header title. That practically means if pop is done in ```SC_CTHREAD``` process, the valid of FIFO is updated at very next clock edge, and will be visible in the process after two cycles. Therefore this ```adv_fifo``` is intended to be used in ```SC_METHOD``` processes only. 

It is easy to update the FIFO for using in ```SC_CTHREAD```  as well. I am planning to provide such implementation near months, but feel free to propose your own implementation.

I have updated the example with method processes.

closed 2022-09-20 by mikhailmoiseev