intel/systemc-compiler #43: segfault when trying to synthesize

Milemarco, 2022-11-14:
Hello, 

I'm trying to convert an existing Open-Source Project to compile it with the ICSC.
I rewrote our test bench for this project and changed the source and header files to be synthesize-able with your tool.

I encountered an error when running the ctest or executing the _sctool directly. while the simulation runs correctly the synthesis throws a Segmentation Fault error.

I tried to identify the error by using the GDB and came to the point in code where the sc_start() gets replaced by the runScElab(__sctool_args_str) for synthesis.

The Segmentation Fault is in the initContext() from libSCTool.so

Do you have any ideas what could possibly be a reason for a segfault here?

Heres the GDB output: 

`Breakpoint 1, sc_main (argc=1, argv=0x5555555ea030) at /home/marco/Workspace/intel_compiler/icsc/designs/paranut_test/dm_tb.cpp:335
335	    int arg, cfg_help = 0;
(gdb) n
338	    arg = 1;
(gdb) 
339	    while (arg < argc && argv[arg][0] == '-') {
(gdb) 
354	    if (cfg_help) {
(gdb) 
363	    sc_clock clk{"clk", sc_time(CLK_PERIOD, SC_NS)};
(gdb) 
364	    Tb tb("tb");
(gdb) 
365	    tb.clk(clk);
(gdb) s
sc_core::sc_in<bool>::operator() (this=0x7fffffff93b0, interface_=warning: RTTI symbol not found for class 'sc_core::sc_clock'
...) at /home/marco/Workspace/intel_compiler/include/sysc/communication/sc_signal_ports.h:495
495		{ this->bind( interface_ ); }
(gdb) n
sc_main (argc=1, argv=0x5555555ea030) at /home/marco/Workspace/intel_compiler/icsc/designs/paranut_test/dm_tb.cpp:368
368	    sc_start ();
(gdb) s
sc::sctool_start<>() () at /home/marco/Workspace/intel_compiler/include/sc_tool/SCTool.h:36
36	    runScElab(__sctool_args_str);
(gdb) 
Finalize elaboration
Design unity file: /home/marco/Workspace/intel_compiler/icsc/designs/paranut_test/build/DebugModule.sctool.cpp

In file included from /home/marco/Workspace/intel_compiler/icsc/designs/paranut_test/build/DebugModule.sctool.cpp:5:
In file included from /home/marco/Workspace/intel_compiler/icsc/designs/paranut_test/dm_tb.cpp:35:
In file included from /home/marco/Workspace/intel_compiler/icsc/designs/paranut_test/dm.h:39:
/home/marco/Workspace/intel_compiler/icsc/designs/paranut_test/base.h:74:2: warning: "INFO: __SYNTHESIS__ is not set! This is just a ParaNut debug info." [-W#warnings]
#warning "INFO: __SYNTHESIS__ is not set! This is just a ParaNut debug info."
 ^
--------------------------------------------------------------
 Intel Compiler for SystemC (ICSC) version 1.4.38, Aug 20,2022
--------------------------------------------------------------
Top-level module is MDebugModule

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff5c7bbcf in sc::ScTraverseConst::initContext() () from /home/marco/Workspace/intel_compiler/lib/libSCTool.so
(gdb) 
Single stepping until exit from function _ZN2sc15ScTraverseConst11initContextEv,
which has no line number information.

Program terminated with signal SIGSEGV, Segmentation fault.
The program no longer exists.
`

mikhailmoiseev, 2022-11-16:
Hello,

There is no enough information to understand the issue.
Could you share the project code with me?
You can do pull request to this repo or give me link to your code. 

-Mikhail.

Milemarco, 2022-11-16:
Hello, 
Thank you for your quick reply.
Here is the link to the Code [https://cloud.hs-augsburg.de/s/kj5Dn3cgbjC79ng](url)

-Marco

mikhailmoiseev, 2022-11-16:
That works for me after a few fixes. Do you mind if I add it into examples? 
If so, you will be able to work on it in this repository and update it as well.

The problem was in CMakeLists.txt --  do not need to create a library, just add all the source files to executable:
```cmake
add_executable(DebugModule dm_tb.cpp dm.cpp base.cpp)
target_include_directories(DebugModule PUBLIC "${PROJECT_SOURCE_DIR}")
svc_target(DebugModule INIT_LOCAL_VARS ELAB_TOP tb.dm)
```

Another fixes:
1. ```PN_WARNINGF``` is not synthesizable as soon as there is pointer initialization, commented 
1. ```wait()``` added before ```while(true)``` in  ```RegisterMethod``` -- suppose that is intended to be reset section

I still see a few warnings related to unsigned/signed types mix and bitwise operation with boolean, but they are not critical.

Milemarco, 2022-11-17:
Great, that worked for me as well. I updated above cloud link so the Code contains necessary license content.
You can add it to the examples, would be a great thing. I still have to rework the `PN_WARNINGF` Macros and i encountered a timing error in the TB Asserts but the synthesis looks fine.
I will look into changing some of the data types to not have these signed/unsigned mixup

This segfault error is hard to debug. I do above restructuring for more Modules and encountered another segfault in a smaller module. Is there a general pattern for running into this error, or what could be a reason for this error

mikhailmoiseev, 2022-11-18:
As soon as you are doing some modification, I suggest you to pull it to ```/examples``` when it is ready (of course if you like). 

Debugging segfault is not easy, it needs to build the ICSC in debug mode. I hope there will be no more such errors.

Cryxos-code, 2022-12-02:
I'm unable to build the ICSC in debug mode.

The installer wont build the ICSC tool inside the `include` folder.
at the near end of installation I get the Error that Cmake cant find sctcommon and failes.
It can't find it because it get's not build when setting the Debug Flag inside the `install.sh` at line 44 `-DCMAKE_BUILD_TYPE=Debug`.
I couldn't figure out what hinders the installer from generating these files because there is no condition or somthing blocking it from doing so.
Is there another way to build it in Debug mode I'm missing?

Milemarco, 2022-12-02:
I'm unable to biuld the ICSC in debug mode.

The installer wont biuld the ICSC tool inside the` include` folder.
at the near end of installation I get the Error that Cmake cant find sctcommon and failes.
It can't find it because it get's not build when setting the Debug Flag inside the `install.sh` at line 44 `-DCMAKE_BUILD_TYPE=Debug`
I couldn't figure out what hinders the installer from generating these files because there is no condition or somthing blocking it from doing so.
Is there another way to build it in Debug mode I'm missing?

mikhailmoiseev, 2022-12-02:
Do you need to have whole ICSC (ScTool.so and others libraries) in debug mode or only your design and SystemC in debug mode (systemc.so)?

Milemarco, 2022-12-02:
I need the whole ICSC in Debug mode if possible

mikhailmoiseev, 2022-12-02:
In this case you need to compile LLVM/Clang and ICSC manually with ```-DCMAKE_BUILD_TYPE=Debug```, see [here](https://github.com/intel/systemc-compiler/wiki/Getting-started#building-and-installing-llvmclang). The ```install.sh``` does not support that.

closed 2022-12-05 by mikhailmoiseev