intel/systemc-compiler #45: support for fied point datatypes sc_fixed sc_ufixed

lbbrhzn, 2022-12-01:
Can you outline what would be needed to add support for sc_fixed and sc_ufixed data types?

I'd be happy to attempt to contribute, but I must admit I am not that familiar with llvm..

mikhailmoiseev, 2022-12-03:
Hello.
You are welcome to contribute into the project. That requires strong C++ experience, lots of attentions to low level details and enough time. That is specially true for arithmetic support, as soon as required to ensure equivalent results between SystemC and generated SV simulations. Please consider >90% of time is testing.

Can I ask you what is your reasons to participate? Are you HW designer who needs fixed/ufixed types?

We are planning to support floating points in Q1`23, and could collaborate in fixed point implementation. As start point you can check ScGenerateExpr::parseBinaryStmt, ScVerilogWriter::putBinary, ScVerilogWriter::getVarDeclVerilog.

-Mikhail.

tvd040562, 2024-01-02:
> Hello. You are welcome to contribute into the project. That requires strong C++ experience, lots of attentions to low level details and enough time. That is specially true for arithmetic support, as [...]

Any solution to fixed point data type?  I currently use scaling to work around the issue.

mikhailmoiseev, 2024-01-02:
No progress in this topic yet. There are some plans to implement it, but priority is low.

closed 2024-08-23 by mikhailmoiseev

lbbrhzn, 2024-08-23:
completed, really?!

mikhailmoiseev, 2024-08-23:
Not completed. No plans to work on that near year, sorry for that.