intel/systemc-compiler #46: Error when building ICSC in Debug mode

Milemarco, 2022-12-09:
I tried to build the ICSC in Debug mode.

I build the protobuf, llvm and clang all in debug mode.
When building ICSC in debug mode i encountered some problems.
I figured out that 
`cmake ../ -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$ICSC_HOME`
doesn't copy the `sctcommon` directory into the include directory. 
This is sovable by copying the directory directly there. 
But then when running `make -j8 VERBOSE=1` i get an error around 61% in the `sct_property.h`

see here:

```
In file included from /data/ICSC_debug/icsc/examples/dvcon20/test_simple.cpp:2:
/data/ICSC_debug/icsc/examples/dvcon20/test_simple.cpp: In member function ‘void Dut<M>::thread_proc()’:
/data/ICSC_debug/include/sctcommon/sct_assert.h:227:50: error: expected identifier before ‘]’ token
  227 |                     [&, SCT_ITER_STR(__VA_ARGS__)]()->bool{return ( LE );},\
      |                                                  ^
/data/ICSC_debug/include/sctcommon/sct_assert.h:266:30: note: in expansion of macro ‘SCT_ASSERT_LOOPN’
  266 | #define SCT_ASSERT_LOOP(...) SCT_ASSERT_LOOPN(__VA_ARGS__)
      |                              ^~~~~~~~~~~~~~~~
/data/ICSC_debug/icsc/examples/dvcon20/test_simple.cpp:70:7: note: in expansion of macro ‘SCT_ASSERT_LOOP’
   70 |       SCT_ASSERT_LOOP(enbl[i], SCT_TIME(1), !enbl[i], i);
      |       ^~~~~~~~~~~~~~~
/data/ICSC_debug/include/sctcommon/sct_assert.h:228:50: error: expected identifier before ‘]’ token
  228 |                     [&, SCT_ITER_STR(__VA_ARGS__)]()->bool{return ( RE );},\
      |                                                  ^
/data/ICSC_debug/include/sctcommon/sct_assert.h:266:30: note: in expansion of macro ‘SCT_ASSERT_LOOPN’
  266 | #define SCT_ASSERT_LOOP(...) SCT_ASSERT_LOOPN(__VA_ARGS__)
      |                              ^~~~~~~~~~~~~~~~
/data/ICSC_debug/icsc/examples/dvcon20/test_simple.cpp:70:7: note: in expansion of macro ‘SCT_ASSERT_LOOP’
   70 |       SCT_ASSERT_LOOP(enbl[i], SCT_TIME(1), !enbl[i], i);
      |       ^~~~~~~~~~~~~~~
/data/ICSC_debug/include/sctcommon/sct_assert.h:233:17: error: expected primary-expression before ‘)’ token
  233 |                 );}
      |                 ^
/data/ICSC_debug/include/sctcommon/sct_assert.h:266:30: note: in expansion of macro ‘SCT_ASSERT_LOOPN’
  266 | #define SCT_ASSERT_LOOP(...) SCT_ASSERT_LOOPN(__VA_ARGS__)
      |                              ^~~~~~~~~~~~~~~~
/data/ICSC_debug/icsc/examples/dvcon20/test_simple.cpp:70:7: note: in expansion of macro ‘SCT_ASSERT_LOOP’
   70 |       SCT_ASSERT_LOOP(enbl[i], SCT_TIME(1), !enbl[i], i);
      |       ^~~~~~~~~~~~~~~

```
and a few lines later

```
/data/ICSC_debug/icsc/examples/dvcon20/test_simple.cpp:72:10:   required from ‘void Dut<M>::thread_proc() [with unsigned int M = 32]’
/data/ICSC_debug/icsc/examples/dvcon20/test_simple.cpp:25:9:   required from ‘Dut<M>::Dut(sc_core::sc_module_name) [with unsigned int M = 32]’
/data/ICSC_debug/icsc/examples/dvcon20/test_simple.cpp:92:23:   required from here
/data/ICSC_debug/include/sctcommon/sct_property.h:368:32: error: no matching function for call to ‘sc_core::sc_spawn_options::set_sensitivity(int*&)’
  368 |             opt.set_sensitivity(event);
      |             ~~~~~~~~~~~~~~~~~~~^~~~~~~
In file included from /data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_method_process.h:46,
                 from /data/ICSC_debug/icsc/systemc/src/systemc:77,
                 from /data/ICSC_debug/icsc/systemc/src/systemc.h:219,
                 from /data/ICSC_debug/icsc/examples/dvcon20/test_simple.cpp:1:
/data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_spawn_options.h:88:10: note: candidate: ‘void sc_core::sc_spawn_options::set_sensitivity(const sc_core::sc_event*)’
   88 |     void set_sensitivity(const sc_event* event)
      |          ^~~~~~~~~~~~~~~
/data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_spawn_options.h:88:42: note:   no known conversion for argument 1 from ‘int*’ to ‘const sc_core::sc_event*’
   88 |     void set_sensitivity(const sc_event* event)
      |                          ~~~~~~~~~~~~~~~~^~~~~
/data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_spawn_options.h:91:10: note: candidate: ‘void sc_core::sc_spawn_options::set_sensitivity(sc_core::sc_port_base*)’
   91 |     void set_sensitivity(sc_port_base* port_base)
      |          ^~~~~~~~~~~~~~~
/data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_spawn_options.h:91:40: note:   no known conversion for argument 1 from ‘int*’ to ‘sc_core::sc_port_base*’
   91 |     void set_sensitivity(sc_port_base* port_base)
      |                          ~~~~~~~~~~~~~~^~~~~~~~~
/data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_spawn_options.h:94:10: note: candidate: ‘void sc_core::sc_spawn_options::set_sensitivity(sc_core::sc_interface*)’
   94 |     void set_sensitivity(sc_interface* interface_p)
      |          ^~~~~~~~~~~~~~~
/data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_spawn_options.h:94:40: note:   no known conversion for argument 1 from ‘int*’ to ‘sc_core::sc_interface*’
   94 |     void set_sensitivity(sc_interface* interface_p)
      |                          ~~~~~~~~~~~~~~^~~~~~~~~~~
/data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_spawn_options.h:97:10: note: candidate: ‘void sc_core::sc_spawn_options::set_sensitivity(sc_core::sc_export_base*)’
   97 |     void set_sensitivity(sc_export_base* export_base)
      |          ^~~~~~~~~~~~~~~
/data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_spawn_options.h:97:42: note:   no known conversion for argument 1 from ‘int*’ to ‘sc_core::sc_export_base*’
   97 |     void set_sensitivity(sc_export_base* export_base)
      |                          ~~~~~~~~~~~~~~~~^~~~~~~~~~~
/data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_spawn_options.h:100:10: note: candidate: ‘void sc_core::sc_spawn_options::set_sensitivity(sc_core::sc_event_finder*)’
  100 |     void set_sensitivity(sc_event_finder* event_finder)
      |          ^~~~~~~~~~~~~~~
/data/ICSC_debug/icsc/systemc/src/sysc/kernel/sc_spawn_options.h:100:43: note:   no known conversion for argument 1 from ‘int*’ to ‘sc_core::sc_event_finder*’
  100 |     void set_sensitivity(sc_event_finder* event_finder)
      |                          ~~~~~~~~~~~~~~~~~^~~~~~~~~~~~

```

I set the `-g` flag inside the `CmakeLists.txt` in the `sctool` directory, that solved the problem for me.

mikhailmoiseev, 2022-12-10:
_I figured out that_
_cmake ../ -DCMAKE_BUILD_TYPE=Release -DCMAKE_INSTALL_PREFIX=$ICSC_HOME_
_doesn't copy the sctcommon directory into the include directory._

If you build debug mode it needs to use ```-DCMAKE_BUILD_TYPE=Debug``` 
Copying of sctcommon should be done by ```make install``` command. Did you run it?

I will update the ```install.sh``` soon to provide debug mode for SystemC library that allows to build user design in debug mode also. In my understanding debug mode for ICSC/Clang/LLVM is required only for ICSC development purpose. Please correct me if I am wrong.

mikhailmoiseev, 2022-12-13:
The ```install.sh``` updated to support Debug and Release mode for user designs. Release mode used by default, it needs to use  ```-DCMAKE_BUILD_TYPE=Debug``` for debug mode.

Milemarco, 2022-12-13:
> _If you build debug mode it needs to use -DCMAKE_BUILD_TYPE=Debug
> Copying of sctcommon should be done by make install command. Did you run it?_

That's what I did.
As in the [getting started](https://github.com/intel/systemc-compiler/wiki/Getting-started#manual-installation) manual installation and in the install.sh described, I first run make before make install.
I encountered the errors while running the make command.

Maybe this is an order of execution problem or did i miss something.
If the copy command would be done earlier i think it would have run.
Please correct me if I screwed something up.

> _In my understanding debug mode for ICSC/Clang/LLVM is required only for ICSC development purpose. Please correct me if I am wrong._

No i think your right, I wanted the sctool build with debug flags to be able to view it with gdb. To be able to debug segfaults.
I think I don't need the Clang/LLVM in debug mode to do so but I wanted to test it anyways.

mikhailmoiseev, 2022-12-13:
Sound good. 
I have merged your pull request, and renamed the DebugModule target to dm_paranut to easily distinguish from other targets. please do git pull.

mikhailmoiseev, 2023-01-07:
Have you translated paranut to Verilog? Any performance/area estimations or issues?

Milemarco, 2023-01-10:
I still have a few modules to modify. This Question is one I'm very interested in. I will let you now as soon as i can make it.

Milemarco, 2023-01-19:
I have tested 3 modules so far. All Modules save around 1-2% of LUTs in comparison to the Vivado-hls.
The synthesis with the icsc is faster by far, compared to Vivado-hls. 
I used Vivado to build the paranut with the verilog modles and it looks fine so far.
Furthermore the ICSC solves a big implementation issue we had when using dynamic memory allocation for submodules.

~ Marco

mikhailmoiseev, 2023-01-20:
Thank you for feedback.

-Mikhail.

Milemarco, 2023-01-20:
In the release Version 1.5.1 the new updated sct_fifo is called in the /design/examples/CmakeLists.txt in line 34: add_subdirectory(sct_fifo)
but there is no such directory inside the examples folder.

Just wan't to let you now.

closed 2023-12-20 by mikhailmoiseev