intel/systemc-compiler #48: Record Type Object errors

Milemarco, 2023-01-19:
I have problems with a Struct i use for data storage.

`struct SWbufEntry{
    sc_uint<30> adr;       
    sc_uint<32> data;
    sc_uint<4> valid;      
    sc_uint<3> special;    
    
    // Necessary operators for using this structure as signals...
    bool operator== (const SWbufEntry &t) const {
        return (adr == t.adr) && (valid == t.valid) && (special == t.special)  && (data == t.data);
    }

    // Displaying
    friend ostream& operator << ( ostream& os, const SWbufEntry &t ) {
        os << "adr:" << t.adr << " data:" << t.data << " valid:" << t.valid;
        return os;
    }
    //Overload function
    friend void sc_trace( sc_trace_file* f, const SWbufEntry& t, const std::string& _s ) {
       sc_trace( f, t.adr, _s + ".adr" );
       sc_trace( f, t.data, _s + ".data" );
       sc_trace( f, t.valid, _s + ".valid" );
       sc_trace( f, t.special, _s + ".special" );
    }
};`

Whenever I try to access the member variables I get following errors:

`Top-level module is MLsu
In file included from /home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/build/LsuModule.sctool.cpp:6:
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:449:28: error: No record object found for member call : ARR_16[1](4) Unknown
            wbuf_dirty0 = (wbuf_var[1].valid != 0) || (wbuf_var[1].special != 0);
                           ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:449:56: error: No record object found for member call : ARR_16[1](4) Unknown
            wbuf_dirty0 = (wbuf_var[1].valid != 0) || (wbuf_var[1].special != 0);
                                                       ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:463:28: error: No record object found for member call : ARR_16[1](4) Unknown
            wbuf_dirty0 = (wbuf_var[1].valid != 0) || (wbuf_var[1].special != 0);
                           ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:463:56: error: No record object found for member call : ARR_16[1](4) Unknown
            wbuf_dirty0 = (wbuf_var[1].valid != 0) || (wbuf_var[1].special != 0);
                                                       ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:467:17: error: No record object found for member call : ARR_16[0](4) Unknown
                wbuf_var[n].adr = wbuf_var[n + 1].adr;
                ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:467:35: error: No record object found for member call : ARR_16[1](4) Unknown
                wbuf_var[n].adr = wbuf_var[n + 1].adr;
                                  ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:468:17: error: No record object found for member call : ARR_16[0](4) Unknown
                wbuf_var[n].data = wbuf_var[n + 1].data;
                ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:468:36: error: No record object found for member call : ARR_16[1](4) Unknown
                wbuf_var[n].data = wbuf_var[n + 1].data;
                                   ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:469:17: error: No record object found for member call : ARR_16[0](4) Unknown
                wbuf_var[n].valid = wbuf_var[n + 1].valid;
                ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:469:37: error: No record object found for member call : ARR_16[1](4) Unknown
                wbuf_var[n].valid = wbuf_var[n + 1].valid;
                                    ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:470:17: error: No record object found for member call : ARR_16[0](4) Unknown
                wbuf_var[n].special = wbuf_var[n + 1].special;
                ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:470:39: error: No record object found for member call : ARR_16[1](4) Unknown
                wbuf_var[n].special = wbuf_var[n + 1].special;
                                      ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:474:13: error: No record object found for member call : ARR_16[3](4) Unknown
            wbuf_var[CFG_LSU_WBUF_SIZE - 1].adr = 0;
            ^
[... 72 lines cut ...]
                            ^
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/paranut-config.h:291:28: note: expanded from macro 'CFG_LSU_WBUF_SIZE'
#define CFG_LSU_WBUF_SIZE (1 << CFG_LSU_WBUF_SIZE_LD)
                           ^
In file included from /home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/build/LsuModule.sctool.cpp:6:
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:131:9: fatal error: Code after function return is prohibited
        return 0;
        ^
------------------------------------------------
  Module number       1
  Process number      2
------------------------------------------------
  General statements  161
  Control statements  53
  Assertions          0
  Wait statements     2
------------------------------------------------
----------------------------------------------------------------
 SystemC-to-Verilog translation, OK 
----------------------------------------------------------------
13 warnings and 17 errors generated.`

I dont know how i can fix this error type. Do you have any advice?

~ Marco

mikhailmoiseev, 2023-01-20:
Could you share the code where wbuf_var is declared and the process code where it is used?

Milemarco, 2023-01-20:
Its both used and declared inside of `proc_clk_lsu()` of lsu.cpp
wbuf_var is a struct containing only `sc_uint<>` members and overloaded operators (see lsu.h). 
I uploaded the files here [https://cloud.hs-augsburg.de/s/qKJK7EXr3sXrmNf]( https://cloud.hs-augsburg.de/s/qKJK7EXr3sXrmNf)
Thank you.
~ marco

mikhailmoiseev, 2023-01-21:
I have taken almost all code from proc_clk_lsu, no error up to now. Will continue to investigate. 
Are you using the last ICSC?

mikhailmoiseev, 2023-01-21:
In the last ICSC your updated design is passed, no record related errors. 
It needs to modify code of ```MLsu::FindWbufHit()``` and ```MLsu::FindEmptyWbufEntry()``` as soon as code after return is not supported. Also please check warnings and remarks reported.

```cpp
int MLsu::FindWbufHit(sc_uint<30> adr)
{
    SWbufEntry entry;
    // Special case for entry 0 - No hit when it will be removed from wbuf this cycle...
    entry = WriteSWbufEntry(0);
    
    if ((entry.valid != 0) && (entry.adr == adr) && (wp_ack_reg == 0))
    {
        return 0;
    } else {
        // look for matching adr in SWbufEntry
        int result = -1;
        for (int n = 1; n < CFG_LSU_WBUF_SIZE; n++){
            entry = WriteSWbufEntry(n);
            if (entry.valid != 0 && entry.adr == adr){ result = n; break; }
        }
        return result;
    }
}

int MLsu::FindEmptyWbufEntry()
{
    SWbufEntry entry;
    int result = -1;
    for (int n = 0; n < CFG_LSU_WBUF_SIZE; n++)
    {
        entry = WriteSWbufEntry(n);
        if (entry.valid == 0 && entry.special == 0) { result = n; break; }
    }
    return result;
}
```

Milemarco, 2023-01-23:
I worked with Version 1.43. I updated my ICSC Version and implemented your changes and the Record type errors are gone.
This helps alot, I still get weird `warning: Read not initialized non-channel variable : adr` for all occurrences of the struct variables.

Also I still get this error: 
```
/home/marco/Workspace/intel_compiler/icsc/designs/lsu_test/lsu.cpp:137:56: fatal error: Code after function return is prohibited
            if (entry.valid != 0 && entry.adr == adr){ return n; break; } //added the break
```
which confuses me.
Maybe this is a installation Problem if you didn't get this error.

mikhailmoiseev, 2023-01-23:
That is correct error -- code after ````return``` is not supported, it needs to rewrite it. See my previous comment.

Milemarco, 2023-01-24:
Yes, I made the necessary change and removed the return from the loop. Now I just need to deal with the warnings and remarks.
Thank you alot!

~ marco

closed 2023-01-24 by Milemarco