intel/systemc-compiler #49: build.sh removal intentional ?

dlmiles, 2023-01-21:
A build.sh was added recently and announced in commit log.
Then a few weeks later is was removed, unannounced.
Was the removal intentional ?

Removal commit:
https://github.com/intel/systemc-compiler/commit/c4dfc805846edada1b95fd8beed294b8c8b375f5

mikhailmoiseev, 2023-01-21:
The script returned with ```install_icsc.sh``` name.

dlmiles, 2023-01-21:
Thank you for your commit and push of 1 hour ago, good to see it return.

Should the HASH BANG be the first line in the file ?  https://github.com/intel/systemc-compiler/blob/e3bee31458bbac09a3da77e8400950145b672e6f/install_icsc.sh#L8

It appears to work, but I don't think the 'set -e' fail fail on first error is working as-is.

mikhailmoiseev, 2023-01-22:
This one is our standard bash script first line. What is the problem? How you suggest to update it?

dlmiles, 2023-01-22:
Move line 8 containing `#!/bin/bash -e` as line 1
Put copyright credits below line 1.
Looks like they were inserted above HASH BANG and may not have been present at all in the previous build.sh file this issue#49 refers to.

`#!` only works as line 1.  I think you know this from your comment.
Right now line 8 is ignored as a comment, the `-e` which causes `set -e` option does not take effect.  The `-e` option is a good thing for a build script to fail fast.

This issue can be closed, I have successfully built SystemC over the past 24h.

Many Thanks for the prompt replies.

mikhailmoiseev, 2023-01-23:
Oh, yes. Thank  you. Will fix it next commit.

closed 2023-01-23 by mikhailmoiseev