intel/systemc-compiler #50: Using nested Record leads to ambigious error

xlef13, 2023-02-01:
Hi,

I was very happy to see you implemented the capability to use structs in channels. I was experimenting with this feature and found, that nesting structs is not fully supported yet.

Seemingly nesting a struct with a single member is supported, adding a second member variable leads to the following error in 
the synthesis step (when calling the _sctool binary). In simulation, nested structs are supported and traced correctly.

```
--------------------------------------------------------------
 Intel Compiler for SystemC (ICSC) version 1.5.5, Jan 27,2023
--------------------------------------------------------------
Top-level module is MMemu
Memu_sctool: /data/icsc_1.5.5/include/llvm/ADT/SmallVector.h:277: T& llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::operator[](llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::size_type) [with T = sc_elab::ObjectView; <template-parameter-1-2> = void; llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::reference = sc_elab::ObjectView&; llvm::SmallVectorTemplateCommon<T, <template-parameter-1-2> >::size_type = long unsigned int]: Assertion `idx < size()' failed.
Abgebrochen
```

Here the code for the structs i wanted to use:

```
struct Srecrec {
    sc_uint<34> z;
    // sc_uint<23> k; //error when uncommented

    bool operator== (const Srecrec& other) {
        return (z == other.z);
        // return (z == other.z, k == other.k);
    }

    friend void sc_trace( sc_trace_file *tf, const Srecrec &t, const std::string &name ) {
        PN_TRACE_R(tf, t, z, name);
        // PN_TRACE_R(tf, t, k, name);
    }

    friend ::std::ostream& operator<< (::std::ostream& os, const Srecrec& s) {
        os << "{" << s.z << "}";
        return os;
    }
};
struct SRec {
    
    int x;
    sc_uint<2> y;
    Srecrec z;

    bool operator== (const SRec& other) {
        return (x == other.x && y == other.y && z == other.z);
    }

    friend void sc_trace( sc_trace_file *tf, const SRec &t, const std::string &name ) {
        PN_TRACE_R(tf, t, y, name);
        PN_TRACE_R(tf, t, x, name);
        PN_TRACE_R(tf, t, z, name);
    }

    friend ::std::ostream& operator<< (::std::ostream& os, const SRec& s) {
        os << "{" << s.x << s.y << "}";
        return os;
    }

};

// PN TRACE MACRO FOR REFERENCE
    #define PN_TRACE_R(TF, OBJ, MEMBER, STR)                        \
        {                                                           \
            if (TF) sc_trace (TF, (OBJ).MEMBER, STR + "."#MEMBER);  \
        }
```

You may recognize the "PN_TRACE_R" from [Milemarco](https://github.com/Milemarco) and the ParaNut, I too am working on that project.

Greetings

mikhailmoiseev, 2023-02-02:
Hi,

Nested records are not supported. There is no plans to support it near future as it is quite difficult.

See all the supported SystemC at https://github.com/intel/systemc-compiler/wiki/SystemC--supported 
"5. Records are supported with limitations. ... Record cannot have another record members."

-Mikhail.

mikhailmoiseev, 2023-03-31:
Added fatal error reporting for nested records.

closed 2023-03-31 by mikhailmoiseev