intel/systemc-compiler #58: sys::CleanupOnSignal error

drom, 2023-08-06:
* OpenSuSe Tumbleweed linux x5 6.4.6-1-default
* Fresh clone
* ./install.sh

Getting this error:

```
In file included from /home/drom/work/github/intel/systemc-compiler/build_deps/llvm-12.0.1.src/lib/Support/Signals.cpp:225:
/home/drom/work/github/intel/systemc-compiler/build_deps/llvm-12.0.1.src/lib/Support/Unix/Signals.inc:348:44: error: ‘void llvm::sys::CleanupOnSignal(uintptr_t)’ should have been declared inside ‘llvm::sys’
  348 | void sys::CleanupOnSignal(uintptr_t Context) {
      |                                            ^
make[2]: *** [lib/Support/CMakeFiles/LLVMSupport.dir/build.make:1882: lib/Support/CMakeFiles/LLVMSupport.dir/Signals.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/Makefile2:9937: lib/Support/CMakeFiles/LLVMSupport.dir/all] Error 2
make: *** [Makefile:156: all] Error 2
```

mikhailmoiseev, 2023-08-08:
Looks like the problem in LLVM installation. Probably it is because of old g++ or std library.

 ```install.sh``` has been checked for Ubuntu 20/22 and SLES 12 only. For other Linux it needs to install Clang/LLVM, Protobuf and ICSC manually, how it is described at https://github.com/intel/systemc-compiler/wiki/Getting-started#manual-installation

drom, 2023-08-10:
Is it new enough? 
```
drom@x5:~/work/github/sifive/sc-rocket-surge> g++ --version
g++ (SUSE Linux) 13.1.1 20230720 [revision 9aac37ab8a7b919a89c6d64bc7107a8436996e93]
Copyright (C) 2023 Free Software Foundation, Inc.
```

drom, 2023-08-10:
How do I check `std` library version?

mikhailmoiseev, 2024-01-05:
CLang/LLVM updated to 15.0.7. Please try it.

closed 2024-08-23 by mikhailmoiseev