intel/systemc-compiler #60: CMake Error at CMakeLists.txt:28 (svc_target):   Unknown CMake command "svc_target".

songlinli0803, 2023-08-08:
Hello，

I have installed icsc, but failed to test the example "counter" with the error below:

CMake Error at CMakeLists.txt:28 (svc_target):
  Unknown CMake command "svc_target".

Does anyone have some suggestions? Thanks.

mikhailmoiseev, 2023-08-08:
Looks like SVC package is not available. Please check 1) you are using bash terminal, 2) run setenv.sh in installation folder in this terminal, 3) version of cmake >= 3.13  and g++ >= 9.0.0.

songlinli0803, 2023-08-09:
Hello mikhailmoiseev，

Thanks for your reply. I have confirmed that : 

$ cmake --version
cmake version 3.22.1

$ g++ --version
g++ (Ubuntu 11.4.0-1ubuntu1~22.04) 11.4.0
Copyright (C) 2021 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ echo $0
-bash

Another thing wizzard is that my installation folder miss 2 files: "CMakeLists.txt" and "README", compared to the installation folder showed in the Userguide. I don't know if it's normal... so i copy the "CMakeLists.txt" from the counter example to the installation folder to do Cmake and the error is like above. By the way, i use WSL Ubuntu 22.04.

songlinli0803, 2023-08-09:
Installation folder like this：
[image]

songlinli0803, 2023-08-09:
This is the log when i just run counter : 
`Start testing: Aug 09 11:44 CST
----------------------------------------------------------
5/1496 Testing: counter_BUILD
5/1496 Test: counter_BUILD
Command: "/usr/bin/cmake" "--build" "/mnt/d/songlinli/SystemC/sc_tools2/build" "--target" "counter_sctool"
Directory: /mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter
"counter_BUILD" start time: Aug 09 11:44 CST
Output:
----------------------------------------------------------
Built target counter_sctool
<end of output>
Test time =   0.50 sec
----------------------------------------------------------
Test Passed.
"counter_BUILD" end time: Aug 09 11:44 CST
"counter_BUILD" time elapsed: 00:00:00
----------------------------------------------------------

6/1496 Testing: counter_SYN
6/1496 Test: counter_SYN
Command: "/mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter/counter_sctool"
Directory: /mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter
"counter_SYN" start time: Aug 09 11:44 CST
Output:
----------------------------------------------------------
/mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter/counter_sctool: error while loading shared libraries: libsc_elab_proto.so: cannot open shared object file: No such file or directory
<end of output>
Test time =   0.02 sec
----------------------------------------------------------
Test Failed.
"counter_SYN" end time: Aug 09 11:44 CST
"counter_SYN" time elapsed: 00:00:00`

songlinli0803, 2023-08-09:
Here is the log information when i run only ctest -R counter:

Start testing: Aug 09 11:44 CST
----------------------------------------------------------
5/1496 Testing: counter_BUILD
5/1496 Test: counter_BUILD
Command: "/usr/bin/cmake" "--build" "/mnt/d/songlinli/SystemC/sc_tools2/build" "--target" "counter_sctool"
Directory: /mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter
"counter_BUILD" start time: Aug 09 11:44 CST
Output:
----------------------------------------------------------
Built target counter_sctool
<end of output>
Test time =   0.50 sec
----------------------------------------------------------
Test Passed.
"counter_BUILD" end time: Aug 09 11:44 CST
"counter_BUILD" time elapsed: 00:00:00
----------------------------------------------------------

6/1496 Testing: counter_SYN
6/1496 Test: counter_SYN
Command: "/mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter/counter_sctool"
Directory: /mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter
"counter_SYN" start time: Aug 09 11:44 CST
Output:
----------------------------------------------------------
/mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter/counter_sctool: error while loading shared libraries: libsc_elab_proto.so: cannot open shared object file: No such file or directory
<end of output>
Test time =   0.02 sec
----------------------------------------------------------
Test Failed.
"counter_SYN" end time: Aug 09 11:44 CST
"counter_SYN" time elapsed: 00:00:00
----------------------------------------------------------

End testing: Aug 09 11:44 CST

songlinli0803, 2023-08-09:
Seems like the error is:

error while loading shared libraries: libsc_elab_proto.so: cannot open shared object file: No such file or directory

But the libsc_elab_proto.so is well placed in the folder "**sc_tools2/lib**" and :

$ echo $LD_LIBRARY_PATH
:/mnt/d/songlinli/SystemC/sc_tools2/lib64:/mnt/d/l00827695/SystemC/**sc_tools2/lib**

Really need help pls...

songlinli0803, 2023-08-09:
If i run only ctest -R counter, the log file is as below where youcould find the error:

----------------------------------------------------------
5/1496 Testing: counter_BUILD
5/1496 Test: counter_BUILD
Command: "/usr/bin/cmake" "--build" "/mnt/d/songlinli/SystemC/sc_tools2/build" "--target" "counter_sctool"
Directory: /mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter
"counter_BUILD" start time: Aug 09 15:37 CST
Output:
----------------------------------------------------------
Consolidate compiler generated dependencies of target counter_sctool
Building CXX object designs/examples/counter/CMakeFiles/counter_sctool.dir/counter.sctool.cpp.o
Linking CXX executable counter_sctool
Built target counter_sctool
<end of output>
Test time =   2.58 sec
----------------------------------------------------------
Test Passed.
"counter_BUILD" end time: Aug 09 15:37 CST
"counter_BUILD" time elapsed: 00:00:02
----------------------------------------------------------

6/1496 Testing: counter_SYN
6/1496 Test: counter_SYN
Command: "/mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter/counter_sctool"
Directory: /mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter
"counter_SYN" start time: Aug 09 15:37 CST
Output:
----------------------------------------------------------

        SystemC 2.3.4-Accellera --- Aug  8 2023 17:03:36
        Copyright (c) 1996-2022 by all Contributors,
        ALL RIGHTS RESERVED
Time		Reset		Counter		Even
Design unity file: /mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter/counter.sctool.cpp

In file included from /mnt/d/songlinli/SystemC/sc_tools2/build/designs/examples/counter/counter.sctool.cpp:1:
In file included from /mnt/d/songlinli/SystemC/sc_tools2/include/sc_tool/SCTool.h:15:
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/string:40:
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/char_traits.h:40:
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/bits/postypes.h:40:
In file included from /usr/lib/gcc/x86_64-linux-gnu/11/../../../../include/c++/11/cwchar:44:
/usr/include/wchar.h:35:10: fatal error: 'stddef.h' file not found
#include <stddef.h>
         ^~~~~~~~~~
Finalize elaboration
--------------------------------------------------------------
 Intel Compiler for SystemC (ICSC) version 1.5.13, Jul 28,2023
--------------------------------------------------------------
Top module is Dut
counter_sctool: /mnt/d/songlinli/SystemC/sc_tools2/llvm-12.0.1.src/tools/clang/lib/AST/RecordLayoutBuilder.cpp:3195: const clang::ASTRecordLayout& clang::ASTContext::getASTRecordLayout(const clang::RecordDecl*) const: Assertion `!D->isInvalidDecl() && "Cannot get layout of invalid decl!"' failed.
<end of output>
Test time =   1.67 sec
----------------------------------------------------------
Test Failed.

Really needs help pls

songlinli0803, 2023-08-09:
If i run only ctest -R counter, the log file is as below where youcould find the error:
[LastTest.log](https://github.com/intel/systemc-compiler/files/12299621/LastTest.log)

mikhailmoiseev, 2023-08-09:
> The problem is solved. In fact, i copy "icsc/cmake/CMakelists.top" to the installation folder and change the name to  "CMakelists.txt", and then run cmake ../ and ctest -j8, all goes well!

That is done by ```install.sh``` script. Did you use it? If so, looks like installation went wrong. I have checked installation with ```instal.sh``` at Ubuntu 22.04 again, it works well.

The current problem from your log is:
```
/usr/include/wchar.h:35:10: fatal error: 'stddef.h' file not found
#include <stddef.h>
```
Probably it needs to add path to STD library headers to $PATH. In my Ubuntu 22.04 it is ```/usr/lib/gcc/x86_64-linux-gnu/11/include```

songlinli0803, 2023-08-10:
> Probably it needs to add path to STD library headers to $PATH. In my Ubuntu 22.04 it is `/usr/lib/gcc/x86_64-linux-gnu/11/include`

Thanks for your advise, now it works for all examples! 

100% tests passed, 0 tests failed out of 50

> That is done by install.sh script. Did you use it? If so, looks like installation went wrong. I have checked installation with instal.sh at Ubuntu 22.04 again, it works well.

Just a feedback: I use WSL Ubuntu 22.04, install.sh doesn't work for me due to serveral problems (so i installed ICSC manually). I give some examples here:

1. `tar -xf v3.13.0.tar.gz --skip-old-files` this command need add -m, if not, this is  a error with 'timestamp' something like this
2. `mkdir build -p && cd build` for protobuf, this command will report file already exist, and i found a file named BUILD, so i change the command to `mkdir build1 -p && cd build1` 
3. I can't 'su' in WSL environment, but some command need add sudo in front.

closed 2023-08-11 by songlinli0803