intel/systemc-compiler #61: Parameter transfer during module instantiation

songlinli0803, 2023-08-26:
Hello,

I define a parameter `NUM` in My_module :  `static const int	NUM = 7;`

And this parameter is for port : `sc_in< sc_uint<NUM> >			a{"a"};`

Then i would like to initialize this parameter in module constructor like this:
`SC_CTOR(My_module );
	My_module (const sc_module_name& name, int NUM):
		sc_module(name), NUM(NUM)
    {}`

But it doesn't work, because of the error : 
` error: 'const int My_module::NUM' is a static data member; it can only be initialized at its definition`

I also tried to change the signal definition to `sc_in<bool>			a[NUM];` but can't work either.

Do you have some good advise for this parameter transfer need, pls?

mikhailmoiseev, 2023-08-26:
Hello,

Static member cannot be initialized in class constructor as it is common for the class type instances. You can initialize ```NUM``` via template parameter defining multiple types:
```
template <int N> 
class A : public sc_module {
    static const int NUM = N;
    sc_vector<sc_in<bool>>   a{"a", NUM};  // better than sc_in<bool> a[NUM]
    ...
};
A<3> a1{"a1"};
A<5> a2{"a2"};   // A<3> and A<5> are different types
```

-Mikhail

songlinli0803, 2023-08-28:
Thanks Mikhail,

Your method works very well！
There is another question as follow : 

> template <int N> 
class A : public sc_module {
    static const int NUM = N;
    constexpr static int	NUM_W= ceil(log2(NUM));
    ...
};
A<3> a1{"a1"};

I would like to add a parameter NUM_W with the type constexpr, but an error is reported:

> error: constexpr variable 'NUM_W' must be initialized by a constant expression
        constexpr static int NUM_W= ceil(log2(QNUM));

I don't know what constant expression means... do you have some idea?

songlinli0803, 2023-08-29:
It seems very useful, thanks a lot

closed 2023-08-29 by songlinli0803