intel/systemc-compiler #63: Cannot get to compile SystemVerilog with a minimal working example

MauroFoti, 2023-10-27:
I'm trying to figure out the build process to obtain an .sv file from the cpp file. For this, I've copied the `dut.h` file from the counter example (`/designs/examples/counter`) and wrote a minimal `CMakeLists.txt` file, that looks like this:

```cmake
cmake_minimum_required(VERSION 3.12)

project(dut)

## SVC package contains ICSC and SystemC libraries
find_package(SVC REQUIRED)

add_executable(dut dut.h)
svc_target(dut ELAB_TOP tb.top)
```

`dut.h` and `CMakeLists.txt` are the only files in the directory `/home/user/icsc_test`. When running `source /opt/icsc/setenv.sh && cmake .` this is the output I get:

```
-- The C compiler identification is GNU 9.4.0
-- The CXX compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- Configuring done
-- Generating done
-- Build files have been written to: /home/user/icsc_test
```

Everything looks fine, but the folder `/home/user/icsc_test/sv_out` is empty. I am not really familiar with CMake, so any suggestion on what I might be missing are very appreciated.

Note that the compiler doesn't produce any .sv file even for the examples design files provided by the repository.

CryxosMarco, 2023-10-28:
Hey, 

you should see a **makefile** in your folder. 
Use `make` to generate the simulation and synthesis execuatable.
The generated files should match the project name u gave in the cmake so in you example `dut` and `dut_sctool`
You can then either run `ctest` or manually execute if you just want to see sv output `./dut_sctool`

This should get you the .sv files in the `sv_out` folder.

For better readability you should use a build folder for all the generatet files. See for Reference [Getting Started](https://github.com/intel/systemc-compiler/wiki/Getting-started#generate-sv-code-for-examples-tests-and-designs)

- Marco

MauroFoti, 2023-11-02:
> Hey,
>
> you should see a **makefile** in your folder. Use `make` to generate the simulation and synthesis execuatable. The generated files should match the project name u gave in the cmake so in you [...]

When I follow the Getting Started guide you mentioned, no file is generated in the sv_out folder when running cmake. Where would I start to figure out why? The documentation doesn't really explain the build process so that's why I was trying to build a single file

mikhailmoiseev, 2023-11-05:
[Generate SV code for one specific example or design](https://github.com/intel/systemc-compiler/wiki/Getting-started#generate-sv-code-for-one-specific-example-or-design)

```bash
$ cd $ICSC_HOME
$ source setenv.sh                   # setup PATH and LD_LIBRARY_PATH
$ cd build   
$ cmake ../                          # prepare Makefiles in Release mode
$ ctest -R DesignTargetName          # compile and run SV generation for DesignTargetName
```
where DesignTargetName is a target name in CMakeLists.txt.

The .sv file is created in ```build/DesignTargetName/sv_out```

closed 2024-08-23 by mikhailmoiseev