intel/systemc-compiler #64: How to get SystemC IR information?

megleo, 2023-11-01:
Hi, 

I am considering how to parse system c syntax and would like to obtain some information about system c programs from syntax analysis, in order to do some optimization work.

So can you help me explain the code structure of your project? Or is there any relevant document describing the structure of the project?
Any help would be greatly appreciated

mikhailmoiseev, 2023-11-02:
Hi,

The compiler is not intended to extract IR. The IR internally extracted at multiple passes to get enough data to generate SV code.
It could take several months of hard work to gather information required for optimizations. 
Considering existing HLS tools which do SC optimization, what is the reson to start this work?

-Mikhail.

megleo, 2023-11-03:
Hi, 

Systemc is actually executed serially. We want to analyze from the AST which processes can operate in parallel, thereby making systemc execute faster.

mikhailmoiseev, 2023-11-05:
SystemC can run process in multiple threads if it is configured with ```enable-pthreads``` options. Usually, that does not speed up simulation because of inter-process interactions which leads to cache synchronizations of cpu cores. 
It is possible to determine groups of processes which have strong dependencies to execute them in the same thread. This approach can help for some designs and does not help for others. 

What kinds of information do you need for your approach? Have you published your approach (then please send a link) or could you outline it?
Are you planning to open source your tool?

-Mikhail.

megleo, 2023-11-07:
Hi,
Thanks for your reply.

We currently have some ideas to determine the execution order between different processes by analyzing the data competition relationship between different processes, and try our best to ensure that threads without data competition can truly execute in parallel.

What kinds of information do you need for your approach? 
I think it should include all the information about the program.

The reason we are doing this work is to validate the ideas of our paper, and if progress goes smoothly, we are certainly willing to open source.

mikhailmoiseev, 2023-11-07:
Suppose, you mean under "data competition" data dependeciens between processes, i.e. if process1 uses some data from process2.  In cycle accurate SC: processes in the same module interacts through sc_signal, processes in different modules interacts through sc_in/sc_out. 
That practically requires to determine:
1. all the sc_in/sc_out/sc_signal`s which are used inside each process,
2. writer process for each sc_in/sc_out/sc_signal.
Are that correct assumptions?

There is another problem with mapping processes to OS threads. Do you have a solution for that?

-Mikhail.

megleo, 2023-12-02:
Hi，
Sorry for reply so late.
Actually, I want to do some work on parallelizing System C: by analyzing the relationships between System C processes, I can confirm that some processes can be parallelized;
Then, by translating it into system verilog code, parallel relationships are reflected.

That why I want to know how to get SystemC ast or IR information.

Thanks

megleo, 2023-12-02:
Hi, 
As is well known, the kernel implementation of SystemC is serial; But Verilog supports parallelism; Is your team prepared to translate the system c source code into a Verilog that can be executed in parallel?

Thanks

megleo, 2023-12-02:
Hi,
Can I continue to develop new features similar to OoO pdes based on your project? 
Thanks

mikhailmoiseev, 2023-12-02:
Hi,
There is AST and CFG constructed by CLang. ICSC provides access to both of them for each process in the design.
You can see work with CFG at https://github.com/intel/systemc-compiler/blob/main/sc_tool/lib/sc_tool/cfg/ScTraverseConst.cpp (see run()) and work with AST at https://github.com/intel/systemc-compiler/blob/main/sc_tool/lib/sc_tool/cfg/ScTraverseCommon.cpp.

Will share some information about internal implementation and structures. The code is self documented, so you need to read and understand comments.

If you can formulate which exact relations you need, I could try to propose some solution. But, you can go by yourself.

We did not work on parallel execution, and generally I do not believe in success. Cycle-accurate designs usually are fine-graned, each process does small piece of work and depends on other process almost every cycle. It is possible to write the code in another style, but it requires additional effort.

-Mikhail.

megleo, 2023-12-03:
Hi,
Thank you very much for your generous reply. I'll take a look at your suggestion first.

Maybe I didn't describe my needs well.

Actually, what I want to do is
Extract information for each process from the code of SystemC, including static and dynamic analysis, and maintain a symbol table for each process (which includes information on each global variable and port).
The symbol table will record whether this process has read and write actions. If two processes do not have an action to write to the same shared variable, it is considered that there is no conflict. In the subsequent translation process in the system verilog, these two processes should be allowed to execute concurrently.
I'm not sure if this idea can be realized and what it entails

Thanks

mikhailmoiseev, 2023-12-03:
In synthesizable SystemC only channels (sc_signal/sc_fifo) can be used for interprocess communication and only one process can write to the channel. Non-channel variable cannot be used for that. ICSC reports error if two processes access a non-channel variable or two processes write to the channel. TLM2 is not synthesizable and not supported by ICSC.

Two processes cannot be run independently if one of the processes takes results from another. 

Information of used/defined variables by processes  (including channels) extracted and used to distinguish 
combinational variables and registers. You can see ```run``` function at https://github.com/intel/systemc-compiler/blob/main/sc_tool/lib/sc_tool/cthread/ScThreadBuilder.cpp

megleo, 2023-12-04:
Thank you very much for your reply,

rseac, 2023-12-31:
@megleo Accessing the AST is possible as suggested by @mikhailmoiseev to extract some of the SystemC behaviour you require.  I had done something similar in a project [systemc-clang](https://github.com/anikau31/systemc-clang). But, this is purely a static approach. It does have an IR, although not as well documented, but it is available. You could adapt some of the code from there and use it with systemc-compiler to get your desired behaviour.  Hope this helps.

closed 2024-08-23 by mikhailmoiseev