intel/systemc-compiler #65: Write and read access to register in different clock domains

fabo9573, 2023-11-08:
Dear all,

i'm a hardware developer with system verilog background and right now trying to use some systemc in combination with your systemc-compiler.
I think i'm having some understanding issue here, and i would be very happy if someone could point me in the right direction.

I'm trying to implement a small ring buffer using **sc_dt::sc_bv<100>bit_memory;** as the memory.
I have two threads now, one writing to the bit_memory (only some bits of the 100) 
in clock domain A and the other thread reading from those bits (again only some bits at once) in the other clock domain B.

When trying to run the systemc compiler I get the following error: 
**error: Multiple processes access non-channel variable : bit_memory**

I don't get why this is an error, as this construct can actually be built in real hardware.

Thank you very much for any input,

Best regards,

Fabian

mikhailmoiseev, 2023-11-08:
Hi Fabian,

Unfortunately, there is no memory concept in SystemC language. If you need to access a variable from two processes it should be a channel type: sc_signal or sc_in/sc_out bound to  sc_signal or other channel. Non-channel variable can be accessed from one process only (otherwise data race can happen), therefore the error reported.

So, you need ```sc_signal<sc_bv<100>>```.

-Mikhail.

fabo9573, 2023-11-10:
Dear Mikhail,

thank you very much for your fast response. This seems to work.

Thank you so much.

closed 2023-11-12 by mikhailmoiseev