intel/systemc-compiler #68: export SystemVerilog modules to multiple files?

pmerz1, 2023-12-08:
Is there a way for ICSC not to write all SystemVerilog modules to one file? 

I think sometimes it would be more convenient to split the modules into several files to make them easier to read.

What do you think?

mikhailmoiseev, 2023-12-09:
No, only one SV file is generated. From my expirience working with one file is more convinient, specially for designs with hundreds of mudules. 
What advantages of multiple files do you see?

mikhailmoiseev, 2023-12-13:
Do you think it is possible to write a script to split one SV file into multiple files by keywords module/endmodule?
I could add a low priority ticket for that, but cannot promise to implement it soon.

pmerz1, 2023-12-13:
I am using a short script doing this as a workaround. 

And I have no problem if you create a low priority ticket for this.

mikhailmoiseev, 2023-12-14:
Sounds good. Will update you when it is completed.

closed 2024-08-24 by mikhailmoiseev