intel/systemc-compiler #71: Confusing line in getting started guide.

sborisov-int, 2024-02-15:
Hi,

For non-Linux users, the next line in the Getting Started guide may be confusing:
`export` ICSC_HOME=/home/user/sc_tools
I'm guessing you meant to replace the word "user" with the actual username. I didn't understand this and copied this line as is. As a result, it took me some time to realize my mistake. I would add an explanation to replace the word "user" with the actual username.

Thanks,
Slava

mikhailmoiseev, 2024-02-17:
The page updated. Thank you for the feedback.

-Mikhail.

closed 2024-02-17 by mikhailmoiseev

sborisov-int, 2024-02-17:
Hi Mikhail,

It's very nice to hear from you.

I'm starting a new project where I may need to convert SystemC to Verilog. I had a long break from using SystemC, so I had to start all over from the beginning. I'm currently trying to install ICSC in a new WSL2 session, but I'm having problems. I will most likely find a way to compile everything, but it would be better if the installation went without problems and human intervention. Perhaps someone on your team could open a new WSL2 session and try to install ICSC?

Regards,
Slava

***@***.***

Intel Israel (74) Limited

This e-mail and any attachments may contain confidential material for
the sole use of the intended recipient(s). Any review or distribution
by others is strictly prohibited. If you are not the intended
recipient, please contact the sender and delete all copies.