intel/systemc-compiler #72: userguide

guojt10, 2024-03-29:
Can userguide add the following information?

1.  What kind of systemc coding can be translated in "assign" statements， for exanple "assign a = b;" ?
2. How does systemc describe bit operations, e.g. assign b[0] = a[0] & a[1]; ?

Looking forward to your reply。

mikhailmoiseev, 2024-03-29:
Answering your questions:
1. Generally, designing in SystemC it make sense to understand which hardware will be created in ASIC/FPGA, not which SV will be generated. To assign one signal to another SC_METHOD is used in SystemC, which is normally translated into always_comb block. SC_METHOD without sensitivity is translated into assign, but right part can be a constant or literal only.
2. Bit operations implemented in SystemC integer types. See sc_uint/sc_biuint in IEEE 1666-2023.

guojt10, 2024-03-30:
It's all settled.

closed 2024-03-30 by guojt10