intel/systemc-compiler #74: Docker / binary releases?

edwardcwang, 2024-07-23:
Hi @mikhailmoiseev! I created a preliminary Dockerfile and Ubuntu 24.04-based release build as a proof-of-concept below. Any interest in extending more official Docker / binary release support? This would make it a lot easier for users if we don't have to compile it from scratch. Thanks so much in advance for your time and consideration!

https://github.com/edwardcwang/intel-systemc-compiler-docker

mikhailmoiseev, 2024-07-23:
Hi @edwardcwang. 
Lets see if the binary distribution you created will be popular. Please consider there is no funding for that.
-Mikhail.

edwardcwang, 2024-07-23:
One idea is to have Github Actions auto-build and publish? GH Actions is free for public repositories like this one.

https://docs.github.com/en/actions/automating-builds-and-tests

mikhailmoiseev, 2024-08-10:
This repository is mirrow of an internal repository. Only stable release are published here. So, it would be nice to have auto-build, but I have lots of another tasks to do related to bug fixes and features.

closed 2025-02-18 by mikhailmoiseev