intel/systemc-compiler #76: Loading Memory

Pelps12, 2024-11-27:
First of all, thank you for this project. I can't begin to say how helpful it's been for my learning journey.

Is there a way to convert memory loading to $readmem in SystemVerilog? Or do I have to do that after compilation?

mikhailmoiseev, 2024-11-28:
```$writemem```/```$readmem``` is non-synthesizable statement in SV, i.e. logic synthesis tool does not support it. You can use this functions in SV testbench. We normally have SystemC testbench where such operations done in C++ style.

Yes, it is possible to insert SV intrinsic with ```__SC_TOOL_VERILOG_MOD__ ``` macro, see Chapter 9.1 in [UserGuide](https://github.com/intel/systemc-compiler/blob/main/doc/ug.pdf). 

To avoid a SystemC code be translated to SV, ```#ifndef __SC_TOOL__ ...  #endif``` could be used.

Pelps12, 2024-12-05:
Yeah. I know it isn't used generally but I want to use it in Vivado which has support for initial in some places like BRAM.

While it is possible to do insert SV intrinsic, it doesn't seem like it's possible to do without completely replacing all the SystemC code.

mikhailmoiseev, 2024-12-06:
I do not see your code, but what you need is separate design and memory loading parts. Only design part can be generated from SystemC. There is no way to make ICSC generates $writemem / $readmem.

https://github.com/intel/systemc-compiler/blob/main/designs/tests/misc/test_verilog_intrinsic3.cpp shows how to generate an empty module to include it from a SV file.

Pelps12, 2024-12-06:
Sorry I didn't present my code.
I was wondering if there was a way for it to not completely replace the SystemC module but add the initial to it.
It's ok if it can't. I'll just generate the SystemVerilog then add it after then make it intrinsic.

```c++
#include "systemc.h"
#include "fstream"
#include "iomanip"
#include "sstream"
#include "vector"
#include "sct_assert.h"

SC_MODULE(RAM)
{
    std::string __SC_TOOL_VERILOG_MOD__ = R"(
        module RAM (
            input logic clk,
            input logic mem_read,
            input logic mem_write,
            input logic rst,
            input logic enable,
            input logic [31:0] addr,
            input logic [31:0] write_data,
            output logic [31:0] mem_data
        );
        initial begin
            $readmemh("hex_memory_file.mem", regs)
        endmodule)";
    sc_in_clk clk;
    sc_in<bool> mem_read;
    sc_in<bool> mem_write;
    sc_in<bool> rst;
    sc_in<bool> enable;
    // Unncessary: Only 9 bits are needed but to make it work
    // with the alu_result and alu_intermmediate_register, 15 bits are used
    sc_in<sc_bv<32>> addr;
    sc_in<sc_bv<32>> write_data;

    sc_signal<sc_bv<32>> memory[512];
    std::string file_name; // Store the file name

    sc_out<sc_bv<32>> mem_data;

    SC_CTOR(RAM);
    // Constructor that accepts the file name
    RAM(sc_module_name name, std::string filename) : sc_module(name), file_name(filename)
    {
        std::cout << "Starting instance of Memory" << std::endl;
        // SC_THREAD(init_func);
        SC_METHOD(main_func);
        sensitive << addr << mem_read << rst << enable << memory;
        SC_THREAD(write_func);
        sensitive << clk.pos();
        async_reset_signal_is(rst, true);
        // SC_THREAD(display_func);
        // sensitive << mem_ready;
    }
    void main_func()
    {

        // wait(addr.value_changed_event() | rst.value_changed_event() | mem_read.value_changed_event() | mem_write.value_changed_event());

        if (mem_read.read() == true && enable.read() == true)
        {
            // std::cout << "Reading RAM: " << memory[addr.read().to_uint()].read().to_string(SC_HEX) << " at Addr: " << addr.read().to_uint() << std::endl;
            mem_data.write(memory[addr.read().to_uint()]);
        }
        sct_assert_latch(mem_data);
    }

    void write_func()
    {
        wait();
        while (true)
        {
            if (mem_write.read() == true && enable.read() == true)
            {

                std::cout << "Writing RAM: " << write_data.read().to_string(SC_HEX) << " at Addr: " << addr.read().to_uint() << std::endl;
                memory[addr.read().to_uint()] = write_data;
            }
            wait();
        }
    }
};
```

mikhailmoiseev, 2024-12-06:
I would create a RAM_wrapper where instantiated the RAM module generated  from SystemC.

```verilog
module RAM_wrapper (
            input logic clk,
            input logic mem_read,
            input logic mem_write,
            input logic rst,
            input logic enable,
            input logic [31:0] addr,
            input logic [31:0] write_data,
            output logic [31:0] mem_data
        );
 
        RAM ram(.*);

        initial begin
            $readmemh("hex_memory_file.mem", ram.memory)
        end
endmodule;
```

Pelps12, 2024-12-06:
I was able to create it based on this idea. Thank you for you time and this project. I'm just getting into synthesizable SystemC so I appreciate the patience.

Thank you.

closed 2024-12-06 by mikhailmoiseev