intel/systemc-compiler #77: Use of Tool outside of Repo

Pelps12, 2024-12-12:
Is there a documented way to use this tool outside of the cloned folder (`designs` folder).

Thank you.

mikhailmoiseev, 2024-12-13:
The simplest way is to add sym link inside of design folder to a folder outside: ```ln -s outside_folder designs/link_name```

KlausKoder, 2024-12-15:
For what it is worth, here is how I do it

I usually install tools in my home dir under `~/usr/<tool>`:

```
# Build for local install

export ICSC_HOME=~/usr/icsc/
export CMAKE_PREFIX_PATH=$ICSC_HOME
export LD_LIBRARY_PATH=$ICSC_HOME/lib64:$ICSC_HOME/lib
export SYSTEMC_HOME=$ICSC_HOME

# Build
mkdir -p runenv/{build_icsc_rel,build_icsc_dbg} && cd runenv/build_icsc_rel
cmake ../..  -DCMAKE_BUILD_TYPE=Release  -DCMAKE_INSTALL_PREFIX=$ICSC_HOME
#cmake ../.. -DCMAKE_BUILD_TYPE=Debug  -DCMAKE_INSTALL_PREFIX=$ICSC_HOME -DCMAKE_DEBUG_POSTFIX=d
make -j32 && make install  && cd -

# Install extras
cp -rl components/common  $ICSC_HOME/designs/single_source/common
cp cmake/CMakeLists.top   $ICSC_HOME/CMakeLists.txt
```

Now you should be able to compile without the repo:

```
# Test
mkdir -p /tmp/icsc-tests && cd /tmp/icsc-tests

. ~/usr/icsc/setenv.sh # set enviroment

cmake $ICSC_HOME  # or directory with your CMakeLists.txt
cd designs/examples/ && ctest -j30
```

closed 2025-02-04 by mikhailmoiseev