intel/systemc-compiler #78: temporal assertions

FSAz, 2025-01-11:
What libraries do I need to add to be able to use temporal assertions?

mikhailmoiseev, 2025-01-12:
You need to include ```sct_common.h``` (https://github.com/intel/systemc-compiler/blob/main/components/common/sctcommon/sct_common.h)  -- that is headers only library.

mikhailmoiseev, 2025-01-13:
There is an example: https://github.com/intel/systemc-compiler/blob/main/designs/examples/asserts/temp_assert.cpp

FSAz, 2025-01-13:
I used this example, but I encountered a series of errors.

FSAz, 2025-01-13:
in this example I just included sct_assert.h

FSAz, 2025-01-13:
Could the issue be caused by my SystemC version?

mikhailmoiseev, 2025-01-13:
Yes, if you use any other version of SystemC that could lead to errors. The simplest way to get SystemC 3.0.0 is to install ICSC as it described at https://github.com/intel/systemc-compiler/wiki/Getting-started

mikhailmoiseev, 2025-01-13:
What is your SystemC version?

FSAz, 2025-01-13:
my systemC version is 2.3.1

mikhailmoiseev, 2025-01-14:
The current verion is 3.0.0. The assertion likely works with 2.3.3 and 2.3.4.

closed 2025-01-16 by mikhailmoiseev