intel/systemc-compiler #82: Use of ICSC in the PicoNut Project

JoHofmann, 2025-04-30:
Hello,

We are a team of university students currently working on a project to develop a RISC-V processor in SystemC. For this purpose, we are utilizing ICSC to translate our SystemC code to SystemVerilog, and we are leveraging the open-source OSS CAD Suite for synthesis.

Our project is called PicoNut, and it is the successor to the ParaNut project, which is already featured in your examples (at least a part of it). You can find our repository here: https://github.com/hsa-ees/piconut.

We wanted to express our appreciation for your compiler.

-Johannes

mikhailmoiseev, 2025-05-02:
Hi Johannes,

That good to know you are using the compiler. Are you planning to build a silicon or use FPGA?
Any plans of comparison or analysis of SystemC  vs conventional Verilog/VHDL methodology?
If you have questions or issues feel free to ask me.

-Mikhail.

JoHofmann, 2025-06-04:
Hi Mikhail,

Sorry for our delayed answer. As a student project we are focusing on FPGA desgins. The project is still work in progress as well.
We are currently not planning for a broad comparison or analysis between those methodologies. The project has a strong focus on simulation and evaluation of architecture design choices.
For simulation we combine our synthesizable hardware modules with pure C++ Code and libraries, e.g. a Qt GUI. This allows us to develop software on system level without actual hardware. With pure Verilog/VHDL this would be more complicated, SystemC fits the needs of our project in terms of simulation speed as well as debug possibilities - especially for us as students.

-Johannes